Azure/azure-iot-connected-factory #2: link on read.me

lucarv, 2017-04-21:
the link in the read.me page that should direct to an explanation of the solution and the "script" takes to a page that only shows the other two pre configured.

barnstee, 2017-04-21:
I know. :-) We're working on this right now. ETA Monday.

SeanStayn, 2017-04-24:
I have found a spelling mistake: 

# Connected factory preconfigured solution

The connected factory preconfigured solution illustrates how you can get started with your Industrie 4.0 digital **trnasformation.** It brings together key Azure IoT services to enable the following features: OPC UA data ingestion, OPC UA server management, rules and actions and Azure Time Series Insights.

hansgschossmann, 2017-06-05:
Both issues are resolved. Typo fixed and the link https://docs.microsoft.com/en-us/azure/iot-suite/iot-suite-getstarted-preconfigured-solutions has now also a walk-through of the Connected factory preconfigured solution.

closed 2017-06-05 by hansgschossmann