Azure/azure-iot-connected-factory #7: Question: Git repository in Azure App-Service-Editor

JuergenKosel, 2017-06-16:
Hello,

the App-Service-Editor allows to initialize a git repository to track the changes in the WebApp of the connected factory.
I thought it could be a good idea to add [this](https://github.com/Azure/azure-iot-connected-factory) github repository (or a clone) as a remote repository within the WebApp. So that I could easily synchronize/update the deployed WebApp in the console of the App-Service-Editor.
But then I have recognized, that only the WebApp subdirectory should be checked out into the wwwroot folder of the deployed connected-factory.

Are there any recommendations, how to track the version-history of the deployed WebApp?
E.g. using git-subtree?

Greetings
Juergen

hansgschossmann, 2017-06-16:
Hi Juergen,
internally we are using VSOs CI capability to deploy the WebApp and thus have a commitment level version information. The Azure ASE does allow to push quick changes to the deployment, but without any good version tracking as far as i know.
Would it help you if we add some info dialog or version info from the version.txt file in the root of the repo to the dashboard?

Thx
Hans

JuergenKosel, 2017-06-16:
Hello Hans,

in the comments of https://docs.microsoft.com/en-us/azure/iot-suite/iot-suite-guidance-on-customizing-preconfigured-solutions there is the hint to use git / github to redeploy a pre-configured solution.
So I was wondering how to do this, if I need only a subdirectory of this connected-factory.

But now, I believe, that the solution with git-subtree would work. (At least, it works for me.)
Maybe this information is useful for some other users...

Greetings
Juergen

hansgschossmann, 2018-08-23:
Close the issue for now.

closed 2018-08-23 by hansgschossmann