Azure/azure-iot-connected-factory #10: Dependency on PowerShell PSCX

msjaosborn, 2017-07-01:
Apparently you need the PSCX 3.2.2 module installed to use the build.ps1 command.  Please document.

barnstee, 2017-07-03:
Thanks for highlighting this! :-)

JuergenKosel, 2017-07-19:
Hello,

I have run into the same issue.
After installing the pscx module with the power Shell command (as Administrator):
 `Install-Module -Name Pscx`
I run into the next missing module:

```
PS D:\Programmierung\azure-iot-connected-factory> ./build.ps1 build
Import-Module : Das angegebene Modul "Posh-SSH" wurde nicht geladen, da in keinem Modulverzeichnis eine gültige
Moduldatei gefunden wurde.
In D:\Programmierung\azure-iot-connected-factory\build.ps1:142 Zeichen:5
+     Import-Module $ModuleName 4> $null
+     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : ResourceUnavailable: (Posh-SSH:String) [Import-Module], FileNotFoundException
    + FullyQualifiedErrorId : Modules_ModuleNotFound,Microsoft.PowerShell.Commands.ImportModuleCommand

CheckModuleVersion : 2017-07-19 16:50:33Z - Posh-SSH module was not found. Please install version 1.7.7 of the module.
In D:\Programmierung\azure-iot-connected-factory\build.ps1:1856 Zeichen:1
+ CheckModuleVersion Posh-SSH $EXPECTED_POSHSSH_MODULE_VERSION
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (:) [Write-Error], WriteErrorException
    + FullyQualifiedErrorId : Microsoft.PowerShell.Commands.WriteErrorException,CheckModuleVersion

2017-07-19 16:50:33Z - Posh-SSH module was not found. Please install version 1.7.7 of the module.
In D:\Programmierung\azure-iot-connected-factory\build.ps1:148 Zeichen:9
+         throw "$(Get-Date –f $TIME_STAMP_FORMAT) - $ModuleName module ...
+         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : OperationStopped: (2017-07-19 16:5... of the module.:String) [], RuntimeException
    + FullyQualifiedErrorId : 2017-07-19 16:50:33Z - Posh-SSH module was not found. Please install version 1.7.7 of the module.

```

JuergenKosel, 2017-07-20:
Hello,

on my machine, I have installed the powershell modules Pscx, Posh-ssh and Invoke-MsBuild (see my commit https://github.com/JuergenKosel/azure-iot-connected-factory/commit/eeaa8cf9640821fce32b2892c42e6c6cafcde915 ).
But this is still not enough: After the powershell script has already used msbuild, it complains, that it can not find msbuild:

```
PS D:\Programmierung\azure-iot-connected-factory> ./build.ps1 build
2017-07-20 09:27:14Z - Name of the deployment is 'local'
AUSFÜHRLICH: 2017-07-20 09:27:15Z - Restoring nuget packages for solution.
MSBuild auto-detection: using msbuild version '15.1.1012.6693' from 'C:\Program Files (x86)\Microsoft Visual Studio\2017
\Professional\MSBuild\15.0\bin'.
Restoring NuGet package bootstrap.less.3.3.7.
Restoring NuGet package Antlr.3.5.0.2.
Restoring NuGet package cldrjs.0.4.4.
Adding package 'bootstrap.less.3.3.7' to folder 'D:\Programmierung\azure-iot-connected-factory\packages'
Adding package 'Antlr.3.5.0.2' to folder 'D:\Programmierung\azure-iot-connected-factory\packages'
Adding package 'cldrjs.0.4.4' to folder 'D:\Programmierung\azure-iot-connected-factory\packages'
Added package 'Antlr.3.5.0.2' to folder 'D:\Programmierung\azure-iot-connected-factory\packages'

snip

  NuGet Config files used:
      C:\Users\qs\AppData\Roaming\NuGet\NuGet.Config
      C:\Program Files (x86)\NuGet\Config\Microsoft.VisualStudio.Offline.config

  Feeds used:
      https://api.nuget.org/v3/index.json
      C:\Program Files (x86)\Microsoft SDKs\NuGetPackages\
AUSFÜHRLICH: 2017-07-20 09:29:32Z - Building Connectedfactory.sln for configuration 'debug'.
msbuild : Die Benennung "msbuild" wurde nicht als Name eines Cmdlet, einer Funktion, einer Skriptdatei oder eines
ausführbaren Programms erkannt. Überprüfen Sie die Schreibweise des Namens, oder ob der Pfad korrekt ist (sofern
enthalten), und wiederholen Sie den Vorgang.
In Zeile:1 Zeichen:1
+ msbuild Connectedfactory.sln /v:m /p:Configuration=debug
+ ~~~~~~~
    + CategoryInfo          : ObjectNotFound: (msbuild:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

AUSFÜHRLICH: 2017-07-20 09:29:33Z - Packaging for configuration 'debug'.
msbuild : Die Benennung "msbuild" wurde nicht als Name eines Cmdlet, einer Funktion, einer Skriptdatei oder eines
ausführbaren Programms erkannt. Überprüfen Sie die Schreibweise des Namens, oder ob der Pfad korrekt ist (sofern
enthalten), und wiederholen Sie den Vorgang.
In Zeile:1 Zeichen:1
+ msbuild D:\Programmierung\azure-iot-connected-factory/WebApp/WebApp.c ...
+ ~~~~~~~
    + CategoryInfo          : ObjectNotFound: (msbuild:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

Cleaning up previously generated packages
Package : 2017-07-20 09:29:33Z - Failed to find WebApp package in directory
'D:\Programmierung\azure-iot-connected-factory/WebApp/obj/debug/package/WebApp.zip'
In D:\Programmierung\azure-iot-connected-factory\build.ps1:1962 Zeichen:5
+     Package
+     ~~~~~~~
    + CategoryInfo          : NotSpecified: (:) [Write-Error], WriteErrorException
    + FullyQualifiedErrorId : Microsoft.PowerShell.Commands.WriteErrorException,Package

Failed to find package for the WebApp.
In D:\Programmierung\azure-iot-connected-factory\build.ps1:1161 Zeichen:9
+         throw "Failed to find package for the WebApp."
+         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : OperationStopped: (Failed to find package for the WebApp.:String) [], RuntimeException
    + FullyQualifiedErrorId : Failed to find package for the WebApp.
```

Is there still another module missing? Or do I need to add msbuild tosome PATH variable?

hansgschossmann, 2017-07-20:
We are going to update the requirements section shortly.

With 1706 Release we switched to VS2017, which means you need to have the right msbuild in you path.

This is what I am using in my $profile to setup the path correct:
# add msbuild
#$env:path = $env:path + ";C:\Program Files (x86)\MSBuild\14.0\bin\"
$env:path = $env:path + ";C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\MSBuild\15.0\Bin"
$env:VisualStudioVersion="15.0"
$env:VSToolsPath="C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\MSBuild\Microsoft\VisualStudio\v15.0"

      https://api.nuget.org/v3/index.json

hansgschossmann, 2017-08-07:
the readme has been updated with the 1707 release. pls check if we missed anything.

closed 2017-08-07 by hansgschossmann