Azure/azure-iot-connected-factory #11: Question: Connect owen OPC UA server fails

J7S, 2017-07-10:
Hello,

I currently try to understand how to connect an own OPC UA server to the connected factory.
Currently I run the edge proxy  and publisher on my development machine.
When trying to connect the opc ua server from the Connected factory dashboard I always get an error message "Error while connecting to the OPC UA Server"

The command shell of the edge proxy prints out the following messages:

(io_decode_message:906) [ERROR] Received message with incompatible version 6.
(io_iot_hub_umqtt_connection_on_receive:407) [ERROR] Failed to decode message (er_invalid_format)

Is there a version mismatch or anything else wrong with my settings?

Thanks!
 Joerg

marcschier, 2017-07-11:
Hi @J7S, This is likely due to a mismatch in version.  If you are building directly from GitHub then you might run into this.  We are in the process of locking down the proxy code for 1.0.0 release and the current master branch in the proxy repo is incompatible with the connected factory, which uses 0.1.x version of the proxy protocol.  So if you use dockerhub images or GitHub please use the 0.1.4.x tag when testing against current connected factory master branch.   We are shooting for an update by end of July.   

If I am incorrect and you are using the dockerhub image with version 0.1.4 (like in the build.ps1 in connected factory), then please reopen.  In that case, please upload a larger snapshot of the log file from proxy to get a better idea as to when the invalid format error occurred.

Thx.

marcschier, 2017-07-11:
Btw., for all 1.x.x releases we aim to keep the clients to be backwards compatible with older proxies.

closed 2017-07-11 by marcschier

J7S, 2017-07-11:
Hi Marc,
yes, your're right. I used the latest proxy version from GitHub. 
I'll switch to the dockerhub image you mentioned.

Thanks!