Azure/azure-iot-connected-factory #12: Connection not reliable

JuergenKosel, 2017-07-17:
Hello,

I am using the proxyd version from https://github.com/Azure/iot-edge-opc-proxy/releases/tag/1.0.0 in my gateway device.
And I have updated the WebApp of the connected factory to the proxy_1_0_0 branch of this project.

In the Browser/index page and the AddOpcServer/index page of my connected factory instance, a valid user could connect to my opc ua servers. But this is not reliable:
After the connection is closed and should be reestablished some time later, the reestablishment fails.
The connection could only be reestablished, if I restart the WebApp of the connected factory.

Greetings Juergen

hansgschossmann, 2017-07-18:
Juergen,
could you log in to the simulation VM and check in /docker/home/Logs the log file of the proxy or use the Simulation/Factory/Get-SimulationLogs.ps1 script to download the logs to your dev workstation and check if there are any errors to see in the proxy log file.

Thx
Hans

JuergenKosel, 2017-07-18:
Unfortunately I can not reproduce this issue anymore. -> Closing this issue.

BTW: I have only updated the WebApp to the proxy_1_0_0 branch with Visual Studio, because:
1. I am more interested in "real" data from my own opc ua server than from the simulation.
2. I do not how to update anything else beside the WebApp. Is there a description how to update the other components of the connected factory?

barnstee, 2017-07-18:
What other components do you want to update? The Proxy and the Publisher?

JuergenKosel, 2017-07-18:
It would be fine to have every thing on the same version. Which should ease debugging or avoid incompatibility issues.
Therefore I want to update also the proxy and the publisher in the VM.

JuergenKosel, 2017-07-19:
My colleague was able to reproduce the issue.
Here the output within the connected factory WebApp:

```
2017-07-19T08:03:03  PID[7084] Error       Exception while trying to connect to the OPC UA server.
Exceptionmessage:
Error establishing a connection: BadNotConnected
Inner Message:
--
Stack trace:
   at Opc.Ua.Bindings.ChannelAsyncOperation`1.End(Int32 timeout, Boolean throwOnError) in d:\a\1\s\Stack\Opc.Ua.Core\Stack\Tcp\ChannelAsyncOperation.cs:line 172
   at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.EndSendRequest(IAsyncResult result) in d:\a\1\s\Stack\Opc.Ua.Core\Stack\Tcp\UaSCBinaryClientChannel.cs:line 303
   at Opc.Ua.Bindings.UaSCUaBinaryTransportChannel.EndSendRequest(IAsyncResult result) in d:\a\1\s\Stack\Opc.Ua.Core\Stack\Tcp\UaSCBinaryTransportChannel.cs:line 334
   at Opc.Ua.Bindings.UaSCUaBinaryTransportChannel.SendRequest(IServiceRequest request) in d:\a\1\s\Stack\Opc.Ua.Core\Stack\Tcp\UaSCBinaryTransportChannel.cs:line 290
   at Opc.Ua.DiscoveryClient.GetEndpoints(RequestHeader requestHeader, String endpointUrl, StringCollection localeIds, StringCollection profileUris, EndpointDescriptionCollection& endpoints) in d:\a\1\s\Stack\Opc.Ua.Core\Stack\Generated\Opc.Ua.Client.cs:line 4821
   at Opc.Ua.DiscoveryClient.GetEndpoints(StringCollection profileUris) in d:\a\1\s\Stack\Opc.Ua.Core\Stack\Client\DiscoveryClient.cs:line 66
   at Opc.Ua.ConfiguredEndpoint.UpdateFromServer(Uri endpointUrl, MessageSecurityMode securityMode, String securityPolicyUri) in d:\a\1\s\Stack\Opc.Ua.Core\Stack\Configuration\ConfiguredEndpoints.cs:line 1040
   at Opc.Ua.ConfiguredEndpoint.UpdateFromServer() in d:\a\1\s\Stack\Opc.Ua.Core\Stack\Configuration\ConfiguredEndpoints.cs:line 1020
   at Opc.Ua.Client.Session.<Create>d__72.MoveNext() in d:\a\1\s\SampleApplications\SDK\Opc.Ua.Client\Session.cs:line 762
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter`1.GetResult()
   at Microsoft.Azure.IoTSuite.Connectedfactory.WebApp.Helpers.OpcSessionHelper.<GetSessionAsync>d__15.MoveNext() in D:\Programmierung\azure-iot-connected-factory\WebApp\Helpers\OpcSessionHelper.cs:line 214
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter`1.GetResult()
   at Microsoft.Azure.IoTSuite.Connectedfactory.WebApp.Controllers.BrowserController.<Connect>d__7.MoveNext() in D:\Programmierung\azure-iot-connected-factory\WebApp\Controllers\BrowserController.cs:line 130
2017-07-19T08:06:58  PID[8784] Error       Object reference not set to an instance of an object.
2017-07-19T08:07:20  PID[8784] Error       Object reference not set to an instance of an object.
2017-07-19T08:08:01  PID[8784] Error       An exception occured while trying to access the OPC UA server.
Exceptionmessage:
BadRequestTimeout
Inner message:
--
Stack trace:
   at Opc.Ua.Bindings.ChannelAsyncOperation`1.End(Int32 timeout, Boolean throwOnError) in d:\a\1\s\Stack\Opc.Ua.Core\Stack\Tcp\ChannelAsyncOperation.cs:line 172
   at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.EndSendRequest(IAsyncResult result) in d:\a\1\s\Stack\Opc.Ua.Core\Stack\Tcp\UaSCBinaryClientChannel.cs:line 303
   at Opc.Ua.Bindings.UaSCUaBinaryTransportChannel.EndSendRequest(IAsyncResult result) in d:\a\1\s\Stack\Opc.Ua.Core\Stack\Tcp\UaSCBinaryTransportChannel.cs:line 334
   at Opc.Ua.Bindings.UaSCUaBinaryTransportChannel.SendRequest(IServiceRequest request) in d:\a\1\s\Stack\Opc.Ua.Core\Stack\Tcp\UaSCBinaryTransportChannel.cs:line 290
   at Opc.Ua.DiscoveryClient.GetEndpoints(RequestHeader requestHeader, String endpointUrl, StringCollection localeIds, StringCollection profileUris, EndpointDescriptionCollection& endpoints) in d:\a\1\s\Stack\Opc.Ua.Core\Stack\Generated\Opc.Ua.Client.cs:line 4821
   at Opc.Ua.DiscoveryClient.GetEndpoints(StringCollection profileUris) in d:\a\1\s\Stack\Opc.Ua.Core\Stack\Client\DiscoveryClient.cs:line 66
   at Opc.Ua.ConfiguredEndpoint.UpdateFromServer(Uri endpointUrl, MessageSecurityMode securityMode, String securityPolicyUri) in d:\a\1\s\Stack\Opc.Ua.Core\Stack\Configuration\ConfiguredEndpoints.cs:line 1040
   at Opc.Ua.ConfiguredEndpoint.UpdateFromServer() in d:\a\1\s\Stack\Opc.Ua.Core\Stack\Configuration\ConfiguredEndpoints.cs:line 1020
   at Opc.Ua.Client.Session.<Create>d__72.MoveNext() in d:\a\1\s\SampleApplications\SDK\Opc.Ua.Client\Session.cs:line 762
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter`1.GetResult()
   at Microsoft.Azure.IoTSuite.Connectedfactory.WebApp.Helpers.OpcSessionHelper.<GetSessionAsync>d__15.MoveNext() in D:\Programmierung\azure-iot-connected-factory\WebApp\Helpers\OpcSessionHelper.cs:line 214
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
[... cut ...]
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter`1.GetResult()
   at Microsoft.Azure.IoTSuite.Connectedfactory.WebApp.Controllers.BrowserController.<Connect>d__7.MoveNext() in D:\Programmierung\azure-iot-connected-factory\WebApp\Controllers\BrowserController.cs:line 130
```
After a restart of the WebApp, the connection to the opc ua server could be established. And the opc ua browsing succeeded.

JuergenKosel, 2017-07-19:
And here the output of proxyd of the embedded gateway device, including the previous good connection:

```
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (pal_getaddrinfo:1176) [INFO] Resolving fd00:42:2001:db8:206:71ff:fe33:14f:4840 (family: 0)...
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (pal_net_trace_resolved:44) [DEBUG]   ... fd00:42:2001:db8:206:71ff:fe33:14f:4840 -> [fd:4200:120:b80d:602:ff71:33fe:4f01]:4840 (fd00:42:2001:db8:206:71ff:fe33:14f)
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (pal_net_trace_resolved:44) [DEBUG]   ... fd00:42:2001:db8:206:71ff:fe33:14f:4840 -> [fd:4200:120:b80d:602:ff71:33fe:4f01]:4840 ()
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (pal_net_trace_resolved:44) [DEBUG]   ... fd00:42:2001:db8:206:71ff:fe33:14f:4840 -> [fd:4200:120:b80d:602:ff71:33fe:4f01]:4840 ()
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (io_iot_hub_umqtt_connection_on_send:328) [INFO] OUT: [PING#424]
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (io_iot_hub_umqtt_connection_on_receive:450) [INFO] IN: [LINK#425]
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (pal_socket_open_by_name:974) [INFO] Resolving fd00:42:2001:db8:206:71ff:fe33:14f:4840...
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (pal_getaddrinfo:1176) [INFO] Resolving fd00:42:2001:db8:206:71ff:fe33:14f:4840 (family: 2)...
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (pal_net_trace_resolved:44) [DEBUG]   ... fd00:42:2001:db8:206:71ff:fe33:14f:4840 -> [fd:4200:120:b80d:602:ff71:33fe:4f01]:4840 (fd00:42:2001:db8:206:71ff:fe33:14f)
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (pal_net_trace_resolved:44) [DEBUG]   ... fd00:42:2001:db8:206:71ff:fe33:14f:4840 -> [fd:4200:120:b80d:602:ff71:33fe:4f01]:4840 ()
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (pal_net_trace_resolved:44) [DEBUG]   ... fd00:42:2001:db8:206:71ff:fe33:14f:4840 -> [fd:4200:120:b80d:602:ff71:33fe:4f01]:4840 ()
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (pal_socket_try_open:849) [DEBUG] Socket created ... (fd:14)
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (pal_socket_connect:705) [DEBUG] Socket connecting ... (fd:14)
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (pal_socket_connect:738) [DEBUG] Socket connected synchronously!
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (pal_socket_try_next:922) [DEBUG] Socket 14 opened synchronously!
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (io_iot_hub_umqtt_connection_on_send:328) [INFO] OUT: [LINK#425]
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (io_iot_hub_umqtt_connection_on_receive:450) [INFO] IN: [OPEN#426]
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (prx_server_socket_handle_openrequest:1473) [INFO] Socket open!
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (io_iot_hub_umqtt_connection_on_send:328) [INFO] OUT: [OPEN#426]
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (io_iot_hub_umqtt_connection_on_receive:445) [DEBUG] IN: [DATA#428]
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (io_iot_hub_umqtt_connection_on_send:323) [DEBUG] OUT: [POLL#428]
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (io_iot_hub_umqtt_connection_on_receive:445) [DEBUG] IN: [POLL#427]
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (io_iot_hub_umqtt_connection_on_send:323) [DEBUG] OUT: [DATA#6263]
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (io_iot_hub_umqtt_connection_on_receive:445) [DEBUG] IN: [DATA#430]
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (prx_server_socket_handle_datamessage:1550) [ERROR] Received bad sequence number, got 0, got 1.
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (io_iot_hub_umqtt_connection_on_send:323) [DEBUG] OUT: [POLL#430]
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (io_iot_hub_umqtt_connection_on_receive:445) [DEBUG] IN: [POLL#429]
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (io_iot_hub_umqtt_connection_on_send:323) [DEBUG] OUT: [DATA#6265]
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (io_iot_hub_umqtt_connection_on_receive:445) [DEBUG] IN: [DATA#432]
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (prx_server_socket_handle_datamessage:1550) [ERROR] Received bad sequence number, got 0, got 1.
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (io_iot_hub_umqtt_connection_on_send:323) [DEBUG] OUT: [POLL#432]
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (io_iot_hub_umqtt_connection_on_receive:445) [DEBUG] IN: [POLL#431]
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (io_iot_hub_umqtt_connection_on_send:323) [DEBUG] OUT: [DATA#6267]
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (io_iot_hub_umqtt_connection_on_receive:445) [DEBUG] IN: [POLL#433]
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (io_iot_hub_umqtt_connection_on_send:323) [DEBUG] OUT: [DATA#6268]
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (io_iot_hub_umqtt_connection_on_receive:445) [DEBUG] IN: [POLL#434]
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (io_iot_hub_umqtt_connection_on_send:323) [DEBUG] OUT: [DATA#6269]
Jul 19 08:05:49 localhost proxyd: Wed Jul 19 08:05:49 2017 (io_iot_hub_umqtt_connection_on_receive:445) [DEBUG] IN: [POLL#435]
[... cut ...]
Jul 19 08:08:31 localhost proxyd: Wed Jul 19 08:07:12 2017 (prx_server_socket_free:282) [INFO] Server socket 0x75411f80 destroyed!
Jul 19 08:08:31 localhost proxyd: Wed Jul 19 08:07:31 2017 (io_mqtt_connection_operation_callback:824) [INFO] Connection alive...
Jul 19 08:08:31 localhost proxyd: Wed Jul 19 08:07:51 2017 (io_mqtt_connection_operation_callback:824) [INFO] Connection alive...
Jul 19 08:08:31 localhost proxyd: Wed Jul 19 08:08:11 2017 (io_mqtt_connection_operation_callback:824) [INFO] Connection alive...
```

marcschier, 2017-07-19:
Looks like an issue in proxy.  Will look at it.

marcschier, 2017-07-19:
I have opened an issue for the duplicate message in the proxy repo:  https://github.com/Azure/iot-edge-opc-proxy/issues/57.   See it for info.  I assume this is all related.

marcschier, 2017-07-22:
Hi, Please try with proxy1_0_2 branch and proxy 1.0.2, which should have a fix for the instability.  Since this issue morphed I suggest to reopen the for instability of browsing, but log a new issue for other items, such as publisher, etc.  Thanks!!!

closed 2017-07-22 by marcschier