Azure/azure-iot-connected-factory #15: Exception while initializing OPC UA stack| Build Using powershell

PedroFloresGuillen, 2017-07-18:
I did a build of connected factory with powershell, and when I whent in the tab of browser/index I found this error:
Exception while initializing OPC UA stack. Error: Access is denied. Inner: -- Stack: at System.Security.Cryptography.X509Certificates.X509Store.Open(OpenFlags flags) at Opc.Ua.X509CertificateStore.Add(X509Certificate2 certificate, String password) in d:\a\1\s\Stack\Opc.Ua.Core\Security\Certificates\X509CertificateStore.cs:line 119 at CertificateFactory.CreateCertificate(String storeType, String storePath, String password, String applicationUri, String applicationName, String subjectName, IList`1 domainNames, UInt16 keySize, DateTime startTime, UInt16 lifetimeInMonths, UInt16 hashSizeInBits, Boolean isCA, X509Certificate2 issuerCAKeyCert) in d:\a\1\s\Stack\Opc.Ua.Core\Security\Certificates\CertificateFactory.cs:line 391 at Opc.Ua.Configuration.ApplicationInstance.d__53.MoveNext() in d:\a\1\s\SampleApplications\SDK\Opc.Ua.Configuration\ApplicationInstance.cs:line 0 --- End of stack trace from previous location where exception was thrown --- at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task) at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task) at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd(Task task) at Opc.Ua.Configuration.ApplicationInstance.d__49.MoveNext() in d:\a\1\s\SampleApplications\SDK\Opc.Ua.Configuration\ApplicationInstance.cs:line 995 --- End of stack trace from previous location where exception was thrown --- at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task) at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task) at System.Runtime.CompilerServices.TaskAwaiter`1.GetResult() at Microsoft.Azure.IoTSuite.Connectedfactory.WebApp.Helpers.OpcSessionHelper.d__9.MoveNext() in C:\Users\PFlores\Desktop\IIOT\Trunk\azure-iot-connected-factory\WebApp\Helpers\OpcSessionHelper.cs:line 89

barnstee, 2017-07-18:
The Web app doesn't have write access to the X509Store it uses, hence access denied. The only reason why the web app would want to write to the X509Store is because it tried to create a self-signed certificate for the OPC UA client built into the web app. However, this should not be necessary as we deploy a self-signed certificate for the web app to use during deployment. Can you check in the Azure portal (portal.azure.com) if you see a cert called "UA Web Client" under SSL certificates (i.e. go to your connected factory resource group, open the App Service and select SSL Certificates)? My hunch is that somehow adding the cert to the web app didn't work during provisioning. Did you get any errors from the script?

PedroFloresGuillen, 2017-07-19:
\azure-iot-connected-factory\WebApp\WebApp.csproj(959,3): error MSB4019: The imported
 project "C:\ProgramFiles\dotnet\sdk\1.0.4\Microsoft\VisualStudio\v10.0\WebApplications\Microsoft.WebApplication.targets" was not found. Confirm that the path in the <Import> declaration is correct, and that the file exists on disk. 

I have redeployed and found this error, also, there is no certificate under SSL certificates that I can see.

barnstee, 2017-07-19:
That's what I thought. So you're missing the cert and this is why the built-in OPC UA browser can't initialize the OPC UA stack. I don't think the missing  Microsoft.WebApplication.targets project is the root-cause, but we'll investigate...

hansgschossmann, 2017-08-07:
The missing .targets file Could you pls give it a try? If you indeed face a missing cert error, then the script Simulation/Factory/Update-Certificate.ps1 could help you. It is deleting the current cert in use, generates a new one and updates it in KeyVault in the Simulation VM and also in your local environement.

hansgschossmann, 2017-08-23:
Have you had a chance to verify if the proposed steps are fixing your issue? If i do not hear anything back within next week, i am going to close the issue.

hansgschossmann, 2017-08-31:
Issue seems to be no longer there, since i have nothing heard back. Please reactivate if you hit it again.

closed 2017-08-31 by hansgschossmann