Azure/azure-iot-connected-factory #18: Does the Proxy have a dependency on the publisher?

ukluk, 2017-07-21:
Does the Proxy “talks” to the Publisher in the edge gateway in order to update the PublishedNode.js file? If the proxy has any dependency to the publisher, we suggest to decouple it.

barnstee, 2017-07-22:
The OPC UA client built into the Connected Factory web portal talks to the Publisher's OPC UA server interface via the Proxy to update the OPC UA pub/sub telemetry messages sent to IoT Hub. The Publisher then updates and persists the publishednodes.json. There is no dependency between the Publisher and the Proxy, they are completely separate.

closed 2017-07-22 by barnstee

ukluk, 2017-07-23:
Erich,

The confusion is if the OPC UA client (portal) sends a message down to the publisher to update the list.

We where under the impression that command and control is proxy and telemetry is publisher... but if portal can send a command to publisher that explains it.

We tried to kill the proxy after browsing for points but publish registration failed. Maybe we did something wrong. Thanks for confirmation.