Azure/azure-iot-connected-factory #20: Unable to create unique name for resource

brownNinja17, 2017-08-23:
(My issue seems similar to [this one](https://github.com/Azure/azure-iot-remote-monitoring/issues/430))

I am trying to deploy the solution on the cloud, and i am getting the following error:

PS C:\Users\Darji\Github\azure> ./build.ps1 cloud -DeploymentName OrangeFactory1324 -Configuration debug
VERBOSE: Cannot verify the Microsoft .NET Framework version 4.5.2 because it is not included in the list of permitted v
VERBOSE: Loading module from path 'C:\Program Files (x86)\Microsoft SDKs\Azure\PowerShell\ServiceManagement\Azure\.\Aut
VERBOSE: Loading module from path 'C:\Program Files (x86)\Microsoft SDKs\Azure\PowerShell\ServiceManagement\Azure\.\Com
VERBOSE: Loading module from path 'C:\Program Files (x86)\Microsoft SDKs\Azure\PowerShell\ServiceManagement\Azure\.\HDI
VERBOSE: Loading module from path 'C:\Program Files (x86)\Microsoft SDKs\Azure\PowerShell\ServiceManagement\Azure\.\Net
VERBOSE: Loading module from path 'C:\Program Files (x86)\Microsoft SDKs\Azure\PowerShell\ServiceManagement\Azure\.\Rec
VERBOSE: Loading module from path 'C:\Program Files (x86)\Microsoft SDKs\Azure\PowerShell\ServiceManagement\Azure\.\Rem
VERBOSE: Loading module from path 'C:\Program Files (x86)\Microsoft SDKs\Azure\PowerShell\ServiceManagement\Azure\.\Ser
VERBOSE: Loading module from path 'C:\Program Files (x86)\Microsoft SDKs\Azure\PowerShell\ServiceManagement\Azure\.\Ser
VERBOSE: Loading module from path 'C:\Program Files (x86)\Microsoft SDKs\Azure\PowerShell\ServiceManagement\Azure\.\Sql
VERBOSE: Loading module from path 'C:\Program Files (x86)\Microsoft SDKs\Azure\PowerShell\ServiceManagement\Azure\.\Sto
VERBOSE: Loading module from path 'C:\Program Files (x86)\Microsoft SDKs\Azure\PowerShell\ServiceManagement\Azure\.\Tra
WARNING: 2017-08-23 13:53:12Z - This script was tested with Posh-SSH module version 1.7.7
WARNING: 2017-08-23 13:53:12Z - Found Posh-SSH module version 2.0.1 installed; continuing, but errors might occur
2017-08-23 13:53:12Z - Name of the deployment is 'orangefactory1324'
VERBOSE: 2017-08-23 13:53:13Z - InitializeDeployment settings for'orangefactory1324'
2017-08-23 13:53:13Z - Using deployment settings filename C:\Users\Darji\Github\azure/orangefactory1324.config.user
2017-08-23 13:53:13Z - Validate Azure account 'varad.darji@br-automation.com'
VERBOSE: 2017-08-23 13:53:14Z - Check for profile file 'C:\Users\Darji\Github\azure/varad.darji@br-automation.com.user'
2017-08-23 13:53:14Z - Use saved profile from 'C:\Users\Darji\Github\azure/varad.darji@br-automation.com.user
2017-08-23 13:53:19Z - Selected Azure subscription Nutzungsbasierte Bezahlung with ID baa2b640-54a8-4a5c-8838-12e261a47
2017-08-23 13:53:19Z - Azure location to use is 'West US'
VERBOSE: 2017-08-23 13:53:20Z - Get resource group name for suiteName 'orangefactory1324' and suiteType 'Connectedfacto
2017-08-23 13:53:21Z - Resourcegroup name is 'orangefactory1324'
VERBOSE: 2017-08-23 13:53:21Z - Check if Azure resource: 'orangefactory1324' (type: 'Microsoft.Storage/storageAccounts'
VERBOSE: 2017-08-23 13:53:22Z - Found the resource. Validating exact naming.
VERBOSE: 2017-08-23 13:53:22Z - Resource with matching resource group name and name found.
VERBOSE: 2017-08-23 13:53:23Z - Check if Azure resource: 'orangefactory1324' (type: 'Microsoft.Devices/iotHubs') exists
VERBOSE: 2017-08-23 13:53:23Z - Found hostname: orangefactory1324.azure-devices.net
VERBOSE: 2017-08-23 13:53:23Z - Found hostname: orangefactory132486a0b.azure-devices.net
-----many similar lines-----
VERBOSE: 2017-08-23 13:54:00Z - Found hostname: orangefactory1324febde.azure-devices.net
VERBOSE: 2017-08-23 13:54:00Z - Found hostname: orangefactory1324986cb.azure-devices.net
GetUniqueResourceName : 2017-08-23 13:54:00Z - Unable to create unique name for resource orangefactory1324 for url azure-devices.net

At C:\Users\Darji\Github\azure\build.ps1:364 char:12
+     return GetUniqueResourceName $resourceBaseName $resourceUrl
+            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (:) [Write-Error], WriteErrorException
    + FullyQualifiedErrorId : Microsoft.PowerShell.Commands.WriteErrorException,GetUniqueResourceName

Unable to create unique name for resource orangefactory1324 for url azure-devices.net
At C:\Users\Darji\Github\azure\build.ps1:383 char:13
+             throw ("Unable to create unique name for resource {0} for ...
+             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : OperationStopped: (Unable to creat...ure-devices.net:String) [], RuntimeException
    + FullyQualifiedErrorId : Unable to create unique name for resource orangefactory1324 for url azure-devices.net

hansgschossmann, 2017-08-31:
I will take a look into this next week when i am back from vacation.
Could you check if there have been a storage account been created at all in resource group orangefactory1324?

brownNinja17, 2017-09-06:
Only storage account was created at that time. Nothing else got deployed.

hansgschossmann, 2017-09-12:
This is an issue with the DNS resolution. We have a fix for it in the works.

2017-08-23 13:53:13Z - Validate Azure account 'varad.darji@br-automation.com<mailto:varad.darji@br-automation.com>'

hansgschossmann, 2017-11-05:
this was due to DNS name responding strategy of Azure and should be fixed in the October 2017 release of Cf. Please reactivate if you still see this issue.

closed 2017-11-05 by hansgschossmann