Azure/azure-iot-connected-factory #21: Deploy with free IoT Hub?

lcao8, 2017-08-31:
Is it possible to deploy the connected factory with a free IoT Hub, F1 instead of S1? Leaving the deployment on overnight would cost over $10 and bulk of the cost is from the IoT Hub. I changed the deployment from 1 S1 hub instead of 3, and didn't make much of a difference.  Deployment with F1 hub resulted in the following error,

"max number of Iot hubs exceeded for sku = Free, max allowed = 1, current = 2"

We are in testing mode and would like to avoid the cost as much as possible.

lcao8, 2017-08-31:
I changed IotHubSkuCapacityUnits to 1 in build.ps1 and ConnectedFactoryMapKey.json.

hansgschossmann, 2017-08-31:
The free SKU only supports 8k messages per day, we used the SKU and #of units to run the simulation well. I have added once the LowCost switch to build.ps1 which was intended to achieve what you want, but still use an S1 and 3 units because of the reason above. You could easily change it, but be aware of the side effects.

closed 2017-11-05 by hansgschossmann