Azure/azure-iot-connected-factory #22: Provide sample script to start proxyd and publisher

JuergenKosel, 2017-09-07:
Hello,
the description to deploy a new gateway for the connected in
https://docs.microsoft.com/en-US/azure/iot-suite/iot-suite-connected-factory-gateway-deployment
is out dated. Especially the version of the proxyd in this description does not match the version of proxyd needed by the connected factory.

Therefore please provide sample scripts to start proxyd and publisher within docker containers.
Maybe these scripts could be provided through the Add Gateway page of the connected factory.

hansgschossmann, 2017-11-05:
Would it not make more sense to add the script/batch files to the proxy/publisher github repositories?
Although i like the idea to show it to the user in the Cf UX.

JuergenKosel, 2017-11-06:
Hello,

the point is, that proxyd and publisher need to match to the connected factory.
With this scripts added to the connected factory, docker (and the user) know which version of proxyd and publisher work with the current connected factory.

JuergenKosel, 2017-12-18:
Maybe a docker-compose file would be a good way to start proxyd and publisher in a more comfortable way. See for example https://github.com/JuergenKosel/gitfab

JuergenKosel, 2017-12-18:
Here is the corresponding publisher log file:
[publisher.log.txt](https://github.com/Azure/azure-iot-connected-factory/files/1568022/publisher.log.txt)

barnstee, 2019-09-27:
The docs have been fixed.

closed 2019-09-27 by barnstee