Azure/azure-iot-connected-factory #26: connected factory addresses wrong publisher to publish node

JuergenKosel, 2017-11-17:
Hello,

we have different network for testing:
Network one: domain name "example.com"
In this network runs a publisher and a proxyd in docker containers and several opc ua servers.
So the connected factory could connted for example in this network to the endpoint
opc.tcp://desktop-mypc.example.com:4890:Configuration1
and it could subscribe opc ua nodes with the publisher endpoint opc.tcp://publisher.example.cm:62222/ua/publisher

We have also another network with a more restrictive firewall and mandatory use of http-proxy.
In this network the opc edge publisher can not connect to the IoT-hub.
So in this network we have another proxyd instance running and different other opc ua servers.
In this network the connected factory could connect to the opc ua endpoint
opc.tcp://[fd00:42:2001:db8:206:71ff:fe33:12]:4840/

If somebode clicks on "subscribe node" for this endpoints, the connected factory reports, that a method call has succeeded. But there is no publisher which could connect to this endpoint.
Taking a look at the output of the publisher of the first network (publisher.example.com): It reports that it should should connect to the endpoint within the unreachable other network.

If then the "unpublish node" method is called to stop the publish which could not work the connected factory (and other opc ua clients) report: "unexpected error". Which is probably an issue of the publisher.

hansgschossmann, 2018-10-22:
Juergen, this should be fixed with August release. But I have not tested with IPv6. Have you tested it?

barnstee, 2019-09-27:
Closed as fixed.

closed 2019-09-27 by barnstee