Azure/azure-iot-connected-factory #28: Question: Microsoft.Rdx.Client not longer supported?

devdeer-alex, 2018-04-30:
The sample utilizes the Microsoft.Rdx.Client NuGet a lot. When I search for this package on NuGet it is unlisted now. [This page](https://libraries.io/nuget/Microsoft.Rdx.Client) states that the package was removed. What would be the replacement to this package?

gasherma, 2018-06-06:
We are using a beta version of the NuGet package, TSI team is working on an RTM version which we will then integrate.

jillcary, 2018-08-09:
I believe the replacement currently is the Time Series Insights REST api:
https://docs.microsoft.com/en-us/azure/time-series-insights/time-series-insights-query-data-csharp

hansgschossmann, 2018-08-23:
The REST API has always been there. We are in discussion with the TSI team. I will update this issue as soon as i have a qualified answer.

hansgschossmann, 2018-10-22:
This will not happen any more in this CY. It is the NuGet package is on their backlog for next year.

barnstee, 2019-09-27:
Correct, the official interface is REST.

closed 2019-09-27 by barnstee