Azure/azure-iot-connected-factory #31: Alert not Showing!?

ryan-ms, 2018-06-10:
Hi,

I deployed the connected factory. And add a new factory in ContosoTopologyDescription.json.
Besides, I added some OpcNodes and set the alarm rule just like the code below.
`{
  "NodeId": "ns=2;s=MTCP_CurrentValueTask.CO2",
  "SymbolicName": "CO2",
  "OpCode": "Avg",
  "Units": "ppm",
  "Maximum": 200
}`

After that, I deployed the website from my visual studio again.
And now, I can received the data from my opcua server successfully. (not from simulator)

But the alarm never shows. I am sure the ppm of CO2 is always higher than 200.

Did I miss something?
Please give me some hints if you have some ideas, thank you.

hansgschossmann, 2018-08-23:
Right now we do not support alerts for the actual data values. We only support alerts for KPIs and OEEs.
This would be a nice feature to add.

hansgschossmann, 2018-10-22:
Nothing heard back. Closing.

closed 2018-10-22 by hansgschossmann