Azure/azure-iot-connected-factory #34: Adding Prosys OPC server, throws "BadCertificateUriInvalid"

humblejay, 2018-07-18:
Adding Prosys OPC simulator to Connected Manufacturing throws following error in App Service log -

2018-07-18T19:25:53  PID[6960] Error       Exception while trying to connect to the OPC UA server.
Exceptionmessage:
BadCertificateUriInvalid

The corresponding Prosys log shows "Application URI (urn:hostname:Contoso:FactorySimulation:UA Web Client) does not match the one in certificate: urn:localhost:Contoso:FactorySimulation:UA Web Client" 
 The hostname above  is that of the machine running Prosys simulator.

The workaround is to set Uri validation "false" in Prosys "settings.xml" file 

This may be a Prosys issue, as the OPC client certificate has no reason to have hostname of the OPC client in its urn: .  It seems the Uri validation in Prosys is wrongly implemented, but any thoughts?

hansgschossmann, 2018-08-23:
This looks indeed as a Prosys bug. I never heard about this before though. Can you pls check with them?

hansgschossmann, 2018-10-22:
Nothing heard back for a while. Closing

closed 2018-10-22 by hansgschossmann

deaddox, 2019-01-18:
Hi,

i am now encountering the same issue. Some OPC servers are configured with certificates that have the incorrect URI. Using the OPC reference stack client i can ignore this error and connect. However when going via the proxy i cant seem to connect.

How do i configure the proxy to ignore URI mismatch errors in the cert validation?

hansgschossmann, 2019-01-18:
It is not a proxy config. It needs to be changed in the WebApp. I thought I have changed the parameter (I think in CreateSession) already. Need to check it.

Outlook for Android<https://aka.ms/ghei36> herunterladen