Azure/azure-iot-connected-factory #35: RdxAuthenticationClientSecret is blank

rajat-barik, 2018-08-24:
RdxAuthenticationClientSecret value is blank. Please guide what this value refers to so that I can manually fill it in the Application settings of webapp.

hansgschossmann, 2018-10-22:
This value is used for the WebApp to authenticate against TimeSeriesInsights. You need to go to your resource group in the Azure Portal, lookup your the App Registration of your CF deployment under "Azure Active Directory -> App registrations", create a new Key and put it as value into the environment of your App Service in the resource group of the WebApp. 
Strange is that it is empty. How have you deployed the CF? Via build script or via www.azureiotsolutions.com? Have you got any errors or warnings?

barnstee, 2019-09-27:
Closed as this has been answered.

closed 2019-09-27 by barnstee