Azure/azure-iot-connected-factory #36: Error while running the build.ps1 script

Shruthi-Patil, 2018-10-07:
I followed the procedure mentioned on the page : https://github.com/Azure/azure-iot-connected-factory
I'm getting the following error :
[image] 
Additional Info:
- Script is successfully built
- Versions of programs are up to date.

hansgschossmann, 2018-10-22:
Please ensure you have the Azure PowerShell Command Line Interface installed (from here: https://azure.microsoft.com/en-us/downloads/). The current CF is tested with the April 2018 version (check in "Add or remove program" of you dev box).

In PowerShell "Find-Module AzureRm.Resources" should report Version 6.6.0 and "Get-Command -Module AzureRm.Resources -Name Get-AzureRmADServicePrincipal" should report Verions 5.5.2

Rikoz02, 2019-09-26:
Same issue, I'd like to know which modules (and versions) do you have installed on your local machine

barnstee, 2019-09-27:
Closed as this has been answered.

closed 2019-09-27 by barnstee