Azure/azure-iot-connected-factory #37: Errors while running the build.ps1

David198, 2018-10-15:
Hi,
when i run the build.ps1 i get the following errors.
```
Invoke-RestMethod : {"odata.error":{"code":"Authorization_RequestDenied","message":{"lang":"en","value":"Insufficient
privileges to complete the operation."}}}
At C:\Users\<myuser>\Documents\TestProjekte\AzureIoTConnectedFactory\build.ps1:883 char:19
+ ...   $result = Invoke-RestMethod -Method "POST" -Uri $uri -Headers @{"Au ...
+                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidOperation: (System.Net.HttpWebRequest:HttpWebRequest) [Invoke-RestMethod], WebExc
   eption
    + FullyQualifiedErrorId : WebCmdletWebResponseException,Microsoft.PowerShell.Commands.InvokeRestMethodCommand
UpdateAadApp : 2018-10-15 10:21:00Z - Unable to create application 'connectedfactorytest-app' with IdentifierUri
'https://connectedfactorytest.azurewebsites.net/connectedfactorytest', returned AppId is null.
At C:\Users\<myuser>\Documents\TestProjekte\AzureIoTConnectedFactory\build.ps1:2345 char:1
+ UpdateAadApp $script:AadTenant
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (:) [Write-Error], WriteErrorException
    + FullyQualifiedErrorId : Microsoft.PowerShell.Commands.WriteErrorException,UpdateAadApp

Unable to create application 'connectedfactorytest-app', returned AppId is null.
At C:\Users\<myuser>\Documents\TestProjekte\AzureIoTConnectedFactory\build.ps1:892 char:13
+             throw ("Unable to create application '{0}', returned AppI ...
+             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : OperationStopped: (Unable to creat... AppId is null.:String) [], RuntimeException
    + FullyQualifiedErrorId : Unable to create application 'connectedfactorytest-app', returned AppId is null.
```

As i read the [documentation ](https://github.com/Azure/azure-iot-connected-factory#authorization_requestdenied-message)the first error can be ignored.

> If you see an message Invoke-RestMethod : {"odata.error":{"code":"Authorization_RequestDenied","message":{"lang":"en","value":"Insufficient privileges to complete the operation."}}} This can be ignored, it is displayed when the user doesn't have administrator privileges to the Active Directory, the Script will continue to execute

But the second error is not mentioned yet. Also in the created resource group only the Storage account is created. Is this a problem with permissions or because of the second error in the script?
When i try to create the solution over [https://www.azureiotsolutions.com/Accelerators#solutions/types/CF](https://www.azureiotsolutions.com/Accelerators#solutions/types/CF) i get the following error message.

> You don't have permission to create Azure Active Directory application. Please request your AAD administrator to enable app registraions in 'user settings' of ADD and also assign you diretory role of 'Global administrator' or any 'Limited administrator'. error code:  Authorization_RequestDenied.

Furthermore when i try to build the WebApp locally the nuget restore throwing errors that:

> Unable to find version '1.0.4' of package 'Microsoft.Azure.Devices.Proxy' 

and 

>Unable to find version '1.0.4' of package 'Microsoft.Azure.Devices.Proxy.Opc.Ua'.

 in my nuget sources and the build does not complete successfully. They should be found normally, shouldn 't they?

Can you please give me a little starting aid what am i doing wrong?

hansgschossmann, 2018-10-22:
Looks like you are not able to create a new Application Registration in your Azure Active Directory, because you have not enough permissions.. Try  out in the portal (portal.azure.com), choose "Azure Active Directory->App registration" and try to add a new registration. If that is not possible, then talk to the administrator of you subscription to grant you more permissions.

barnstee, 2019-09-27:
Closed as this has been answered.

closed 2019-09-27 by barnstee