Azure/azure-iot-connected-factory #41: Error on build - wrong package path

danielrmendes, 2018-12-06:
Hello,
Followed the instructions and all seems fine. However when I try to build I get the following error, replicated to all packages:

`Could not find a part of the path 'C:\Users\thisUser\Documents\azure-iot-connected-factory\ \packages\System.Collections.Specialized.4.3.0'.`

The packages exist and are located at:
`Could not find a part of the path 'C:\Users\thisUser\Documents\azure-iot-connected-factory\packages\System.Collections.Specialized.4.3.0'.`
That is, without the extra "\ \".
Been wrapping my head around what is causing this but cannot find a solution.

palyi, 2018-12-06:
I could reproduce the same error with the latest release. All package paths get a "space+\", corrupting each package reference at build.  See screenshot.

AzoricoTC, 2018-12-17:
We managed a temporary fix for this by removing the "\" in "$(MSBuildProjectDirectory)\packages" at lines 39, 40 e 44.

studentpra, 2019-01-09:
Severity	Code	Description	Project	File	Line	Suppression State
Error		Could not find a part of the path 'D:\azure-iot-connected-factory\ \packages\jQuery.UI.i18n.1.10.2'.	WebApp		

hi i am having error above realted to path as its having double \\ before package applied above solution as mentioned in comment also.

Please, help me with issue so that i can build and do some changes in to it.

AzoricoTC, 2019-01-09:
@studentpra 

Hey there go to the .nuget folder, open the Nuget.targets file and remove the extra "\"  on the aforementioned lines. Should fix all those path errors.

studentpra, 2019-01-18:
Hi @AzoricoTC ,

Is there any way to show time series insights data to power bi dashboard 

i tried to search a lot but not able to connect power bi to time series insights 
Please, let me know the solution if any 

thanks
Prateek

sandervandevelde, 2019-02-02:
@AzoricoTC your fix works, thanks. Remove in total 4 times a slash in NuGet.targets.

barnstee, 2019-09-27:
Closed as this has been asnwered.

closed 2019-09-27 by barnstee