Azure/azure-iot-connected-factory #45: Topology changes not being reflected in the app, keeps showing default setup.

Grewall, 2019-03-01:
Hi - I deployed the connected factory solution from azureiotsolutions.com and have been able to successfully create a gateway device that is reporting data in connected factory json format. As a result of that i see a 'New Factory' as one of the factories and i can see my nodes configured on the OPC Publisher. Using directions from [here](https://docs.microsoft.com/en-us/azure/iot-accelerators/iot-accelerators-connected-factory-configure) I changed the ContosoTopologyDescription.json file in folder WebApp>Contoso>Topology using Visual Studio Code and Published the file and restarted the app but i do not see any changes on the Topolog file being reflected on the app and I see the Demo data + the New Factory. 
Do you guys see any obvious issues with the way I am doing it. Please let me know if there are questions i can answer.
Thanks!

hansgschossmann, 2019-03-14:
what you did sounds right. can you provide the .json with your modifications?

barnstee, 2019-09-27:
Closed as this has been answered.

closed 2019-09-27 by barnstee