Azure/azure-iot-connected-factory #50: Time Series Insight, OEE and KPI not works when deployed by build.ps1

ryan-ms, 2019-05-17:
Hi

I just clone the latest repo(with OPC twin integrated), and use ./build.ps1 to provision the connected factory, and it sucessfully deployed.

After I deployed, I did not do anything, but I found that time series insight is not working, so chart of OEE, KPI is also not working. 

My Connected Factoty Picture
[image]

Error log when open Time series insight line chart panel
[image]

[image]

Could you check the code? or did I do something wrong?
Looking forward for your reply, thank you.

ryan-ms, 2019-06-26:
After using www.azureiotsolutions.com to deploying the solution, it's not happen again, I think it can close now.

closed 2019-06-26 by dacolgit

ni3git, 2020-07-17:
I am facing this issue too. Any pointers or resolution?