Azure/azure-iot-connected-factory #62: ASPNETCOMPILER : error ASPRUNTIME: '/temp' is not a valid IIS application.

jjgriff93, 2019-08-27:
When running the build.ps1 script, I get the following error:

ASPNETCOMPILER : error ASPRUNTIME: '/temp' is not a valid IIS application. [C:\Users\jamesgr\desktop\azure-iot-connecte
d-factory\WebApp\WebApp.csproj]

I've made no changes to the source code and followed the instructions closely, not sure what's wrong at a guess the web project within the VS solution doesn't seem to be able to locate the proper virtual directory.

marcschier, 2019-10-20:
Can you provide us with the log output of the script?  Likely an issue with path separators.

johnadali, 2020-04-02:
Hello,
   I am having the same issue as jjgriff93. I have followed the instructions exactly, except I am using Visual Studio 2019 Enterprise, so I had to modify the environment variables. I am attaching the log file that I received when running the build.ps1 script file, along with a screenshot of the environment variables in my Powershell session.

I have also made no changes to the source code, so I'm not sure why I am getting this error.

Please advise.

Thanks.
[connfactory.log](https://github.com/Azure/azure-iot-connected-factory/files/4418475/connfactory.log)
[image]
[image]

dacolgit, 2020-07-22:
Take a look here:
https://stackoverflow.com/questions/37779818/mvcbuildviews-true-causes-temp-is-not-a-valid-iis-application-error
https://developercommunity.visualstudio.com/content/problem/94528/when-setting-true-errror-aspruntime-temp-is-not-a.html
It's a Visual Studio issue.

closed 2020-07-22 by dacolgit