Azure/azure-iot-connected-factory #77: Time Series Insight, OEE and KPI dashboards are not working when deployed using build.ps1

ni3git, 2020-07-17:
[image]
 
Any pointers or resolution?

ni3git, 2020-07-19:
It works when deployed from https://www.azureiotsolutions.com/Accelerators but wanted to make it fully up and running with working dashboards and KPI data when deployed using ps1 file. Any help will be highly appreciated?

ni3git, 2020-07-20:
Below is the error I see in the application log stream:

"2020-07-20T00:50:57 PID[4656] Information RDX CreateQueryClient failed: AADSTS7000215: Invalid client secret is provided.Trace ID: 34c284d3-8e58-4a84-95e7-820392af3b00Correlation ID: 2b751893-a025-405e-9093-a6924064c166Timestamp: 2020-07-20 00:50:57Z---> Response status code does not indicate success: 401 (Unauthorized).---> {"error":"invalid_client","error_description":"AADSTS7000215: Invalid client secret is provided.\r\nTrace ID: 34c284d3-8e58-4a84-95e7-820392af3b00\r\nCorrelation ID: 2b751893-a025-405e-9093-a6924064c166\r\nTimestamp: 2020-07-20 00:50:57Z","error_codes":[7000215],"timestamp":"2020-07-20 00:50:57Z","trace_id":"34c284d3-8e58-4a84-95e7-820392af3b00","correlation_id":"2b751893-a025-405e-9093-a6924064c166","error_uri":"https://login.microsoftonline.com/error?code=7000215"}: Unknown error"

dacolgit, 2020-07-22:
It seems your deployment is corrupted. Please delete and redeploy.

ni3git, 2020-07-22:
Thank you @dacolgit for the response. Issue has been resolved. App service was unable to authenticate against time series insights because of the the invalid token. When I double checked ps1 deployment logs, found out that Remove-AzureRmADAppCredential and New-AzureRmADAppCredential commands were failed. So executed these commands manually using powershell and added randomly generated secret to the app (in AD -> App Registration)

closed 2020-07-22 by ni3git